NVIDIA/gdrcopy #278: Minimal steps to install gdrdrv driver only please

hassanbabaie, 2023-09-26:
Hi there, it's not clear to me what are the minimal steps to only install the driver on an kubernetes node that has only the NVIDIA driver installed.

I'm looking through the issues and it would be really good if this can be broken up on the readme, and if you can let me know here too.

Thanks in advance

hassanbabaie, 2023-09-26:
I just installed the following, on the host

```
yum install -y https://developer.download.nvidia.com/compute/redist/gdrcopy/CUDA%2012.2/rhel8/x64/gdrcopy-kmod-2.4-1dkms.el8.noarch.rpm
yum install -y https://developer.download.nvidia.com/compute/redist/gdrcopy/CUDA%2012.2/rhel8/x64/gdrcopy-devel-2.4-1.el8.noarch.rpm
yum install -y https://developer.download.nvidia.com/compute/redist/gdrcopy/CUDA%2012.2/rhel8/x64/gdrcopy-2.4-1.el8.x86_64.rpm
```

and then ran:

```
# gdrcopy_sanity
Total: 28, Passed: 28, Failed: 0, Waived: 0

```
Does that mean it looks good?

I don't see any files in 

```
# ls -ls /dev/gdrdrv
0 crw-rw-rw- 1 root root 242, 0 Sep 26 01:26 /dev/gdrdrv
```

pakmarkthub, 2023-09-26:
Hi @hassanbabaie,

If you want to install gdrdrv only, `https://developer.download.nvidia.com/compute/redist/gdrcopy/CUDA%2012.2/rhel8/x64/gdrcopy-kmod-2.4-1dkms.el8.noarch.rpm` is sufficient.

```
# ls -ls /dev/gdrdrv
0 crw-rw-rw- 1 root root 242, 0 Sep 26 01:26 /dev/gdrdrv
```
Doesn't it show `/dev/gdrdrv` here?

pakmarkthub, 2023-09-27:
Yes, gdrdrv is now ready on your system. Now, you just need to mount `/dev/gdrdrv`. If you use docker, `docker run <other options> --device=/dev/gdrdrv:/dev/gdrdrv`.

closed 2023-11-22 by pakmarkthub