NVIDIA/gdrcopy #282: How to understand the file "nv-p2p-dummpy.c"

pyt-hnu, 2023-11-09:
What is the purpose of the file "nv-p2p-dummy.c"?
why not to use the symbol of nvidia.ko?

pakmarkthub, 2023-11-22:
Hi @pyt-hnu,

In the current design, we do not force users to install nvidia.ko before compiling gdrdrv.ko. For example, you can just have the nvidia header files ready without installing the nvidia driver. You can compile gdrdrv by `make NVIDIA_SRC_DIR=<path-to-nvidia-header>`. That is a reason why we have nv-p2p-dummy.

pyt-hnu, 2023-11-22:
> Hi @pyt-hnu,
>
> In the current design, we do not force users to install nvidia.ko before compiling gdrdrv.ko. For example, you can just have the nvidia header files ready without installing the nvidia [...]

Thanks for your reply, Does this apply to all GPUDirect RDMA technologies?

pakmarkthub, 2023-11-22:
No. Some drivers are in-tree with nvidia.ko or come with the same installation package as nvidia.ko. So, they have a better way of handling this issue. On the other hand, GDRCopy is off-tree with nvidia.ko. So, we need some workaround.

closed 2023-11-24 by pakmarkthub