NVIDIA/gdrcopy #292: How to effectively test gdrcopy is enable using Real world ML workload ?

pandyamarut, 2024-01-31:
I installed gdrcopy successfully on my host. Successfully ran tests as well. Post installation, I started a container,  which has my LLM application, I am actually profiling model loading from local disk. Are there any adequate methods to test this?

How can I verify that, when I run actual application, gdrcopy is enabled.

pakmarkthub, 2024-02-05:
Hi @pandyamarut,
Based on your question, my guess is that your application does not use GDRCopy directly. Probably you want to confirm that a library (e.g., UCX, NCCL) is properly utilizing GDRCopy? One way to do so is to export the environment variables below and rerun your application. If GDRCopy is used, you should see some output lines from GDRCopy.
```
export GDRCOPY_ENABLE_LOGGING=1
export GDRCOPY_LOG_LEVEL=1
```

dhayanesh, 2024-04-23:
@pandyamarut where you able to verify it?