NVIDIA/gdrcopy #295: Can't make with Intel Compiler

mendax1234, 2024-03-29:
Hi, I try to install `gdrcopy` with the following command
``` bash
make CUDA=/usr/local/modules/cuda-12.3 CC=icc CXX=icpc
make install
```
My icc version is 2021.9.0 and my icpc version is 2021.9.0
I am pretty sure these two intel compiler works fine.
But I still got error like below
```bash
./include/asm-generic/fixmap.h(38): error: asm blocks not enabled, use -fasm-blocks
        BUG_ON(vaddr >= FIXADDR_TOP || vaddr < FIXADDR_START);
        ^

In file included from /home/wenbo/gdrcopy/src/gdrdrv/nv-p2p-dummy.c(45):
./include/linux/mm.h(479): warning #2160: anonymous union qualifier is ignored
        const struct {
        ^

In file included from /home/wenbo/gdrcopy/src/gdrdrv/nv-p2p-dummy.c(45):
./include/linux/mm.h(1182): error: asm blocks not enabled, use -fasm-blocks
        if (WARN_ON_ONCE(page_ref_count(page) <= 0))
```
The whole error is too long, I just choose some of them.

So, how to fix this error. Or, gdrcopy only supports the using GCC to build?

Thank you very much!

pakmarkthub, 2024-03-30:
Hi @mendax1234,

GDRCopy composes of a user-space library and a Linux driver. Normally, you compile your linux kernel with gcc. For compatibility, it is suggested to use the same compiler to also compile Linux drivers. The errors you posted are all related to header files from your Linux kernel. Although I believe that there is a way to make it work, that will require some effort. And you may run into an issue later during runtime if your kernel is not compiled with the Intel compiler. 

You have an option to compile the driver and the user-space library / applications separately. For example, `make lib` will compile the library only. Please look at Makefile for more info.

mendax1234, 2024-03-30:
> Hi @mendax1234,
>
> GDRCopy composes of a user-space library and a Linux driver. Normally, you compile your linux kernel with gcc. For compatibility, it is suggested to use the same compiler to also [...]

Thanks for your  reply! @pakmarkthub 

So, if I use the GCC to compile the drivers and the lib, (as you mentioned there may be some error during the runtime if my kernel is compiled with GCC), will I run into some error when using this gcc compiled gdrcopy when installing some other libraries, like `openmpi`, cuz I want to use `--with-gdrcopy` when building the openmpi with Intel Compiler. But if my gdrcopy is using GCC compiler, I don't know whether there will be some issues.

Thanks!

pakmarkthub, 2024-03-30:
It should work in theory. But we have never tested with Intel compilers.

mendax1234, 2024-03-30:
> It should work in theory. But we have never tested with Intel compilers.

Okay. Sure. Thanks so much!

closed 2024-03-30 by mendax1234