NVIDIA/gdrcopy #298: Assertion "(cuStreamQuery(0)) == (CUDA_ERROR_NOT_READY)"

osayamenja, 2024-06-28:
Running `gdrcopy_pplat` fails with `Assertion "(cuStreamQuery(0)) == (CUDA_ERROR_NOT_READY)" failed at pplat.cu:257`.

See complete logs below
<details>
  <summary>Click me</summary>
  
  ```Bash
GPU id:0; name: Tesla V100-SXM2-32GB; Bus id: 0001:00:00
GPU id:1; name: Tesla V100-SXM2-32GB; Bus id: 0002:00:00
GPU id:2; name: Tesla V100-SXM2-32GB; Bus id: 0003:00:00
GPU id:3; name: Tesla V100-SXM2-32GB; Bus id: 0004:00:00
GPU id:4; name: Tesla V100-SXM2-32GB; Bus id: 0005:00:00
GPU id:5; name: Tesla V100-SXM2-32GB; Bus id: 0006:00:00
GPU id:6; name: Tesla V100-SXM2-32GB; Bus id: 0007:00:00
GPU id:7; name: Tesla V100-SXM2-32GB; Bus id: 0008:00:00
selecting device 0
device ptr: 0x7f6bb7a00000
gpu alloc fn: cuMemAlloc
map_d_ptr: 0x7f6bdc010000
info.va: 7f6bb7a00000
info.mapped_size: 65536
info.page_size: 65536
info.mapped: 1
info.wc_mapping: 1
page offset: 0
user-space pointer: 0x7f6bdc010000
CPU does gdr_copy_to_mapping and GPU writes back via cuMemHostAlloc'd buffer.
Running 1000 iterations with data size 4 bytes.
Assertion "(cuStreamQuery(0)) == (CUDA_ERROR_NOT_READY)" failed at pplat.cu:257
  ```
</details>

If useful, I have [GPUDirectAsync](https://github.com/gpudirect/libgdsync) configured and nvidia-peermem activated.

osayamenja, 2024-06-28:
Every other test works fine. Results are attached below.

<details>
  <summary>gdrcopy_sanity</summary>
  
  ```Bash
Total: 28, Passed: 28, Failed: 0, Waived: 0
  ```
</details>

<details>
  <summary>gdrcopy_copybw</summary>
  
  ```Bash
GPU id:0; name: Tesla V100-SXM2-32GB; Bus id: 0001:00:00
GPU id:1; name: Tesla V100-SXM2-32GB; Bus id: 0002:00:00
GPU id:2; name: Tesla V100-SXM2-32GB; Bus id: 0003:00:00
GPU id:3; name: Tesla V100-SXM2-32GB; Bus id: 0004:00:00
GPU id:4; name: Tesla V100-SXM2-32GB; Bus id: 0005:00:00
GPU id:5; name: Tesla V100-SXM2-32GB; Bus id: 0006:00:00
GPU id:6; name: Tesla V100-SXM2-32GB; Bus id: 0007:00:00
GPU id:7; name: Tesla V100-SXM2-32GB; Bus id: 0008:00:00
selecting device 0
testing size: 131072
rounded size: 131072
gpu alloc fn: cuMemAlloc
device ptr: 7f8293a00000
map_d_ptr: 0x7f82b2423000
info.va: 7f8293a00000
info.mapped_size: 131072
info.page_size: 65536
info.mapped: 1
info.wc_mapping: 1
page offset: 0
user-space pointer:0x7f82b2423000
writing test, size=131072 offset=0 num_iters=10000
write BW: 8680.15MB/s
reading test, size=131072 offset=0 num_iters=100
read BW: 379.824MB/s
unmapping buffer
unpinning buffer
closing gdrdrv
  ```
</details>

<details>
  <summary>gdrcopy_copylat</summary>
  
  ```Bash
GPU id:0; name: Tesla V100-SXM2-32GB; Bus id: 0001:00:00
GPU id:1; name: Tesla V100-SXM2-32GB; Bus id: 0002:00:00
GPU id:2; name: Tesla V100-SXM2-32GB; Bus id: 0003:00:00
GPU id:3; name: Tesla V100-SXM2-32GB; Bus id: 0004:00:00
GPU id:4; name: Tesla V100-SXM2-32GB; Bus id: 0005:00:00
GPU id:5; name: Tesla V100-SXM2-32GB; Bus id: 0006:00:00
GPU id:6; name: Tesla V100-SXM2-32GB; Bus id: 0007:00:00
GPU id:7; name: Tesla V100-SXM2-32GB; Bus id: 0008:00:00
selecting device 0
device ptr: 0x7f68b4000000
allocated size: 16777216
gpu alloc fn: cuMemAlloc

map_d_ptr: 0x7f68e1000000
info.va: 7f68b4000000
info.mapped_size: 16777216
info.page_size: 65536
info.mapped: 1
info.wc_mapping: 1
page offset: 0
user-space pointer: 0x7f68e1000000
[... 182 lines cut ...]
[1482.816000	-	1853.520000]	6
[1853.520000	-	2224.224000]	0
[2224.224000	-	2594.928000]	2
[2594.928000	-	2965.632000]	0
[2965.632000	-	3336.336000]	0
[3336.336000	-	3707.040000]	0
[3707.040000	-	4077.744000]	0

Size(B)	pin.Time(us)	map.Time(us)	get_info.Time(us)	unmap.Time(us)	unpin.Time(us)
16777216	396.702820	25.480310	0.573010	54.088660	209.703560
Histogram of gdr_pin_buffer latency for 16777216 bytes
[379.205000	-	758.410000]	72
[758.410000	-	1137.615000]	20
[1137.615000	-	1516.820000]	5
[1516.820000	-	1896.025000]	1
[1896.025000	-	2275.230000]	1
[2275.230000	-	2654.435000]	0
[2654.435000	-	3033.640000]	0
[3033.640000	-	3412.845000]	0
[3412.845000	-	3792.050000]	0
[3792.050000	-	4171.255000]	0

closing gdrdrv
  ```
</details>

pakmarkthub, 2024-06-28:
Hi @osayamenja,

Did you have `NVCCFLAGS` or `CPPFLAGS` set to something related to `nvcc`. This issue means that the CUDA kernel is launched incorrectly. In most cases, it is from a missing CUDA binary. For example, your NVCCFLAGS might have `--gpu-code=sm_90`. The CUDA binary that you compile out can be run on Hopper but not on Volta. You can try the compile flag below and rerun the new binary again.

```
make clean
NVCCFLAGS="--gpu-architecture=compute_70 --gpu-code=compute_70" make -j4
```

osayamenja, 2024-06-28:
Hey @pakmarkthub thanks for the quick response! I installed gdrcopy using the deb packages, so I am not using `make` to run the tests; I run the installed binary. That said, I tried rerunning `gdrcopy_pplat` with the NVCCFLAGS exported but the error persists. Let me know if you need any extra information.

pakmarkthub, 2024-06-28:
If you use the deb packages, the binary should be compiled with the correct flags. Some requests / questions:
1. The gdrcopy_test package should specify the CUDA version it was compiled with. Please make sure that that CUDA version and the CUDA library on your system are compatible.
2. Can you make sure that you can run a simple CUDA kernel on your system? You can use this https://github.com/NVIDIA/cuda-samples. Please try Samples/0_Introduction/vectorAdd (compile and run it).

osayamenja, 2024-06-28:
1. Running `gdrcopy_test` returns `gdrcopy_test: command not found`
2. Running the CUDA example you suggested works fine, see results below. I also want to mention, this may not be a high-priority issue (yet), since the downstream application I am using, NVSHMEM, works fine with gdrcopy so far.

 <details>
  <summary>vectorAdd</summary>
  
  ```Bash
[Vector addition of 50000 elements]
Copy input data from the host memory to the CUDA device
CUDA kernel launch with 196 blocks of 256 threads
Copy output data from the CUDA device to the host memory
Test PASSED
Done
  ```
</details>

osayamenja, 2024-06-28:
Just in case you need these.

 <details>
  <summary>nvcc --version</summary>
  
  ```Bash
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2024 NVIDIA Corporation
Built on Wed_Apr_17_19:19:55_PDT_2024
Cuda compilation tools, release 12.5, V12.5.40
Build cuda_12.5.r12.5/compiler.34177558_0
  ```
</details>

 <details>
  <summary>nvidia-smi</summary>
  
  ```Bash
Fri Jun 28 03:08:08 2024       
+---------------------------------------------------------------------------------------+
| NVIDIA-SMI 535.183.01             Driver Version: 535.183.01   CUDA Version: 12.2     |
|-----------------------------------------+----------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |         Memory-Usage | GPU-Util  Compute M. |
|                                         |                      |               MIG M. |
|=========================================+======================+======================|
|   0  Tesla V100-SXM2-32GB           On  | 00000001:00:00.0 Off |                    0 |
| N/A   37C    P0              43W / 300W |    103MiB / 32768MiB |      0%      Default |
|                                         |                      |                  N/A |
+-----------------------------------------+----------------------+----------------------+
|   1  Tesla V100-SXM2-32GB           On  | 00000002:00:00.0 Off |                    0 |
| N/A   41C    P0              43W / 300W |      5MiB / 32768MiB |      0%      Default |
|                                         |                      |                  N/A |
+-----------------------------------------+----------------------+----------------------+
|   2  Tesla V100-SXM2-32GB           On  | 00000003:00:00.0 Off |                    0 |
| N/A   38C    P0              42W / 300W |      5MiB / 32768MiB |      0%      Default |
|                                         |                      |                  N/A |
+-----------------------------------------+----------------------+----------------------+
|   3  Tesla V100-SXM2-32GB           On  | 00000004:00:00.0 Off |                    0 |
| N/A   39C    P0              42W / 300W |      5MiB / 32768MiB |      0%      Default |
|                                         |                      |                  N/A |
+-----------------------------------------+----------------------+----------------------+
|   4  Tesla V100-SXM2-32GB           On  | 00000005:00:00.0 Off |                    0 |
| N/A   37C    P0              41W / 300W |      5MiB / 32768MiB |      0%      Default |
|                                         |                      |                  N/A |
+-----------------------------------------+----------------------+----------------------+
|   5  Tesla V100-SXM2-32GB           On  | 00000006:00:00.0 Off |                    0 |
| N/A   39C    P0              42W / 300W |      5MiB / 32768MiB |      0%      Default |
|                                         |                      |                  N/A |
+-----------------------------------------+----------------------+----------------------+
|   6  Tesla V100-SXM2-32GB           On  | 00000007:00:00.0 Off |                    0 |
| N/A   38C    P0              42W / 300W |      5MiB / 32768MiB |      0%      Default |
|                                         |                      |                  N/A |
+-----------------------------------------+----------------------+----------------------+
|   7  Tesla V100-SXM2-32GB           On  | 00000008:00:00.0 Off |                    0 |
| N/A   41C    P0              42W / 300W |      5MiB / 32768MiB |      0%      Default |
|                                         |                      |                  N/A |
+-----------------------------------------+----------------------+----------------------+
                                                                                         
+---------------------------------------------------------------------------------------+
| Processes:                                                                            |
|  GPU   GI   CI        PID   Type   Process name                            GPU Memory |
|        ID   ID                                                             Usage      |
|=======================================================================================|
|    0   N/A  N/A      1848      G   /usr/lib/xorg/Xorg                           33MiB |
|    1   N/A  N/A      1848      G   /usr/lib/xorg/Xorg                            4MiB |
|    2   N/A  N/A      1848      G   /usr/lib/xorg/Xorg                            4MiB |
|    3   N/A  N/A      1848      G   /usr/lib/xorg/Xorg                            4MiB |
|    4   N/A  N/A      1848      G   /usr/lib/xorg/Xorg                            4MiB |
|    5   N/A  N/A      1848      G   /usr/lib/xorg/Xorg                            4MiB |
|    6   N/A  N/A      1848      G   /usr/lib/xorg/Xorg                            4MiB |
|    7   N/A  N/A      1848      G   /usr/lib/xorg/Xorg                            4MiB |
+---------------------------------------------------------------------------------------+
  ```
</details>

pakmarkthub, 2024-06-28:
Thank you for the additional information. GDRCopy does not rely on CUDA. `gdrcopy_pplat` is just a benchmark application. It needs to launch a CUDA kernel because we are measuring a ping-pong latency between CPU and GPU.

Can you try compiling from source? You can still use libgdrapi.so as well as the gdrdrv driver from the deb packages. What you need to compile is just the test applications, but it might be easier to compile the whole project.

> Running gdrcopy_test returns gdrcopy_test: command not found

No. I meant the gdrcopy-test deb package you are using. For example, you probably downloaded something like https://developer.download.nvidia.com/compute/redist/gdrcopy/CUDA%2012.2/ubuntu20_04/x64/gdrcopy-tests_2.4-1_amd64.Ubuntu20_04+cuda12.2.deb. This is a gdrcopy-test deb package compiled with CUDA12.2 on Ubuntu 20.04 for x86-64. This is just an example. Can you check the CUDA version that the gdrcopy-tests deb package you are using was compiled with?

osayamenja, 2024-06-28:
I installed gdrcopy following the README instructions, meaning the script automatically detected my CUDA toolkit and ubuntu version, which is 20.04. I will try recompiling from source and get back soon, thanks for your effort and quick response!