NVIDIA/gdrcopy #300: How should I implement device-to-device memory copying between different hosts?  For example, copying the contents from host C's B GPU memory to host A's CPU memory.

JOjoker-world, 2024-07-17:


pakmarkthub, 2024-07-17:
Hi @JOjoker-world,

This is internode communication. You can use a communication library such as MPI, NCCL, NVSHMEM, etc depending on the programming model your application is using.

JOjoker-world, 2024-07-19:
Thank you very much for your reply. Do you mean that gdrcopy can achieve direct copying between GPU and CPU across different nodes (e.g., A.GPU -> B.CPU) by using the appropriate communication library?

pakmarkthub, 2024-07-19:
No. GDRCopy is an intranode communication library between GPU and CPU. It cannot do internode communication. On the other hand, MPI, NCCL, NVSHMEM are designed to support internode communication. You will want to use them.

JOjoker-world, 2024-07-19:
Alright, I understand. Thank you for your reply.

closed 2024-07-19 by JOjoker-world