NVIDIA/gdrcopy #305: "gdrcopy_sanity" failing with Nvidia 560 drivers on  A100

rafsalas19, 2024-09-19:
OS: Ubuntu 20.04
Kernel: 5.15.0-1071
Nv drivers: 560.35.03 
CUDA: 12.5
gdrcopy: 2.4.1

error:
```
gdrcopy_sanity
Assertion "(gdr_pin_buffer(g, d_A[0], buffer_size, 0, 0, &A_mh[0])) == (0)" failed at sanity.cpp:435
Assertion "(gdr_pin_buffer(g, d_A, A_size, 0, 0, &A_mh)) == (0)" failed at sanity.cpp:344
Total: 28, Passed: 24, Failed: 2, Waived: 2
```

pakmarkthub, 2024-09-20:
Hi @rafsalas19,

This looks like an issue we have already fixed in the master branch. May I ask you to try it out? You will need to compile and install the gdrdrv driver to get this fix.

rafsalas19, 2024-09-20:
Ok thanks let me try

rafsalas19, 2024-09-23:
Ok this worked Thanks! Can you let me know when will there be a release/tag published that has this fix in it?

LiquidPT, 2024-10-15:
Following up on this. We only pull releases to include on our systems. When can we expect a release to include this, it appears, 7 month old fix?

drossetti, 2024-11-18:
@LiquidPT FYI we recently tagged rel 2.4.2.