NVIDIA/gdrcopy #309: Does not compile on RHEL 9.5

bsteinb, 2024-11-17:
The kernel in RHEL 9.5 comes with a backport of https://github.com/torvalds/linux/commit/bc292ab00f6c7a661a8a605c714e8a148f629ef6 which means `mm.h` now defines `vm_flags_set`. However, the [version check in `gdrdrv.c`](https://github.com/NVIDIA/gdrcopy/blob/7e3797ff381844b46fa120b67c6bbcc4e89ab741/src/gdrdrv/gdrdrv.c#L104) does not catch this, as the RHEL 9.5 kernel identifies as `5.14.0`. Building `gdrcopy` thus fails with:

```
  CC [M]  /builddir/build/BUILD/gdrcopy-2.4.1/src/gdrdrv/gdrdrv.o
/builddir/build/BUILD/gdrcopy-2.4.1/src/gdrdrv/gdrdrv.c:102:20: error: redefinition of 'vm_flags_set'
  102 | static inline void vm_flags_set(struct vm_area_struct *vma, vm_flags_t flags)
      |                    ^~~~~~~~~~~~
In file included from /builddir/build/BUILD/gdrcopy-2.4.1/src/gdrdrv/gdrdrv.c:34:
./include/linux/mm.h:826:20: note: previous definition of 'vm_flags_set' with type 'void(struct vm_area_struct *, vm_flags_t)' {aka 'void(struct vm_area_struct *, long unsigned int)'}
  826 | static inline void vm_flags_set(struct vm_area_struct *vma,
      |                    ^~~~~~~~~~~~
/builddir/build/BUILD/gdrcopy-2.4.1/src/gdrdrv/gdrdrv.c: In function 'vm_flags_set':
/builddir/build/BUILD/gdrcopy-2.4.1/src/gdrdrv/gdrdrv.c:104:19: error: assignment of read-only member 'vm_flags'
  104 |     vma->vm_flags |= flags;
      |                   ^~
make[3]: *** [scripts/Makefile.build:249: /builddir/build/BUILD/gdrcopy-2.4.1/src/gdrdrv/gdrdrv.o] Error 1
```

stannous, 2024-12-02:
With gdrcopy 2.4.2, in addition to vm_flags errors on RHEL 9.5, I'm seeing an error about missing file:

/usr/src/nvidia-545.23.08/nvidia/nv-frontend.c: No such file or directory

This is with both latest 545-dkms and 565-dkms GPU drivers.

Linux kernel is 5.14.0-503.15.1.el9_5.x86_64 and cuda-toolkit-12-6 is installed.

```
$ cat /var/lib/dkms/gdrdrv/2.4.2/build/make.log 
DKMS make.log for gdrdrv-2.4.2 for kernel 5.14.0-503.15.1.el9_5.x86_64 (x86_64)
Mon Dec  2 12:52:24 PM PST 2024
make: Entering directory '/usr/src/kernels/5.14.0-503.15.1.el9_5.x86_64'
grep: /usr/src/nvidia-545.23.08/nvidia/nv-frontend.c: No such file or directory
  CC [M]  /var/lib/dkms/gdrdrv/2.4.2/build/nv-p2p-dummy.o
  CC [M]  /var/lib/dkms/gdrdrv/2.4.2/build/gdrdrv.o
/var/lib/dkms/gdrdrv/2.4.2/build/gdrdrv.c:109:20: error: redefinition of ‘vm_flags_set’
  109 | static inline void vm_flags_set(struct vm_area_struct *vma, vm_flags_t flags)
      |                    ^~~~~~~~~~~~
In file included from /var/lib/dkms/gdrdrv/2.4.2/build/gdrdrv.c:34:
./include/linux/mm.h:826:20: note: previous definition of ‘vm_flags_set’ with type ‘void(struct vm_area_struct *, vm_flags_t)’ {aka ‘void(struct vm_area_struct *, long unsigned int)’}
  826 | static inline void vm_flags_set(struct vm_area_struct *vma,
      |                    ^~~~~~~~~~~~
/var/lib/dkms/gdrdrv/2.4.2/build/gdrdrv.c: In function ‘vm_flags_set’:
/var/lib/dkms/gdrdrv/2.4.2/build/gdrdrv.c:111:19: error: assignment of read-only member ‘vm_flags’
  111 |     vma->vm_flags |= flags;
      |                   ^~
make[1]: *** [scripts/Makefile.build:249: /var/lib/dkms/gdrdrv/2.4.2/build/gdrdrv.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:1944: /var/lib/dkms/gdrdrv/2.4.2/build] Error 2
make: Leaving directory '/usr/src/kernels/5.14.0-503.15.1.el9_5.x86_64'
[dgxuser@host-10-34-3-136 ~]$ uname -a
Linux host-10-34-3-136 5.14.0-503.15.1.el9_5.x86_64 #1 SMP PREEMPT_DYNAMIC Thu Nov 14 15:45:31 EST 2024 x86_64 x86_64 x86_64 GNU/Linux
[dgxuser@host-10-34-3-136 ~]$ 
```

drossetti, 2024-12-02:
A fix for both issues is already available and will be pushed to master as well as to v2.4.3 (new minor update release, still under testing, expected very soon).

bsteinb, 2024-12-03:
> With gdrcopy 2.4.2, in addition to vm_flags errors on RHEL 9.5, I'm seeing an error about missing file:
> 
> /usr/src/nvidia-545.23.08/nvidia/nv-frontend.c: No such file or directory

That is not related to RHEL 9.5, but is an incompatibility between `gdrcopy` and driver versions 545 and newer that I reported as #283 back in November 2023. A fix for that was added to the master branch in #284 shortly after, but it has not been included in either the 2.4.1 release made in December 2023 or the 2.4.2 release from this November.

pakmarkthub, 2024-12-03:
> That is not related to RHEL 9.5, but is an incompatibility between `gdrcopy` and driver versions 545 and newer that I reported as [#283](https://github.com/NVIDIA/gdrcopy/issues/283) back in [...]

We also noticed this missing fix in R2.4 branch. As @drossetti said, that fix + RHEL9.5 support will come with v2.4.3 release.

bsteinb, 2024-12-10:
Fixed in release v2.4.3.

closed 2024-12-10 by bsteinb