NVIDIA/gdrcopy #311: scripts/test_gdrdrv_HAVE_VM_FLAGS_SET.sh should honour user-defined kernel source location

bsteinb, 2024-12-10:
The Makefile for the kernel driver, `src/gdrdrv/Makefile`, allows the user to define a custom location for the kernel sources via the `KDIR` variable. However, the test script `scripts/test_gdrdrv_HAVE_VM_FLAGS_SET.sh` ignores the setting and requires the kernel sources to be located in `/lib/modules/${kver}/build`. This is confusing and can lead to inconsistent results.

bsteinb, 2025-04-04:
Same is true for `scripts/test_gdrdrv_HAVE_PROC_OPS.sh` added in version 2.5.

efegokdemir, 2026-09-21:
I’m working on this issue. I’ll update both kernel compatibility probes to honor the existing KDIR override and verify the Makefile passes it through.