NVIDIA/gdrcopy #317: Create conda package for GDRCopy

billysuh7, 2025-02-21:
Holoscan (https://github.com/nvidia-holoscan/holoscan-sdk) depends on part of GDRCopy, i.e.`libgdrapi.so`. It would be great to have GDRCopy (or at least `libgdrapi.so` portion of it) generate the conda package for it.

pakmarkthub, 2025-02-25:
Thank you, @billysuh7. conda is not in our support list. I will discuss with the team about your request and our future direction.

By the way, how do you manage today? Do you install GDRCopy separately since it is not available in conda?

jakirkham, 2025-02-26:
Does the GDRCopy team already provide `redist`s? We could start with that

See [some binaries]( https://developer.download.nvidia.com/compute/redist/gdrcopy/ ). Though the last one is CUDA 12.2

Is it possible to install GDRCopy strictly in user-space? Or are there kernel modules or other pieces needed?

billysuh7, 2025-04-02:
> Does the GDRCopy team already provide `redist`s? We could start with that
>
> See [some binaries](https://developer.download.nvidia.com/compute/redist/gdrcopy/). Though the last one is CUDA 12.2
>
> Is it [...]

Turns out, holoscan no longer needs GDRCopy as a dependency - it was needed by 3rdparty library being revended (UCX) by holoscan, but since conda does not allow it, once we have excluded these files, and the error went away.