NVIDIA/gdrcopy #319: [Question] Cannot install the deb file, any one can help?

vincentccc, 2025-02-28:
[image]

m-antonov, 2025-03-06:
Check the contents of `/var/lib/dkms/gdrdrv/2.5/build/make.log`. I can assume that the contents may be something like following:

```
DKMS make.log for gdrdrv-2.5 for kernel 6.1.0-21-amd64 (amd64)
Thu Mar  6 06:38:34 PM MSK 2025
/usr/sbin/dkms: line 88: cd: /var/lib/dkms/gdrdrv/2.5-1/build: No such file or directory
```

This can be fixed in the `packages/dkms.conf` file by changing the `@FULL_VERSION@` variable to `@VERSION@`. Make the changes before building.

pakmarkthub, 2025-03-06:
Thank you for the report. We have pushed the fix (along with other changes) to the master branch.

pakmarkthub, 2025-03-06:
Using the latest release version can reduce the risk of running into issues. We do QA for all releases. The master branch, on the other hand, is more dynamic and does not go through the QA release cycle.

vincentccc, 2025-03-07:
> Check the contents of `/var/lib/dkms/gdrdrv/2.5/build/make.log`. I can assume that the contents may be something like following:
>
> ```
> DKMS make.log for gdrdrv-2.5 for kernel 6.1.0-21-amd64 (amd64) [...]

That's it! Thanks very much!

vincentccc, 2025-03-07:
> Using the latest release version can reduce the risk of running into issues. We do QA for all releases. The master branch, on the other hand, is more dynamic and does not go through the QA release cycle.

Got that, thx for your reply!