NVIDIA/gdrcopy #323: [Question] How to install gdrcopy in docker container?

zigzagcai, 2025-04-01:
For some reasons, I don't have access to the host machine. 
So, could gdrcopy be installed within docker?

wanglecheng123, 2025-09-02:
@pakmarkthub ，hi, I found the same issue. Does this words -- “You need to be root to install this driver." mean that in container, if in sudo mode, I can install the driver?  If so, can you provide some instructions on how to install the driver?

pakmarkthub, 2025-09-02:
Hi @wanglecheng123,
Do you plan to use the gdrdrv container image? In that case, I suggest that you use it as part of the GPU Operator. That will do the right thing for you. If you want to do it manually, you can use this command as an example `docker run -d --name gdrcopy-driver --privileged -v /:/driver-root --env NVIDIA_DRIVER_ROOT=/driver-root <the-gdrdrv-image>`

closed 2026-10-12 by zigzagcai