NVIDIA/gdrcopy #327: Version compatibility between host driver and installed container image library

eicherseiji, 2025-07-07:
Hi GDRCopy maintainers,

I was unable to find an answer to this via search--

Since GDRCopy needs to be installed in both [the host and container image](https://github.com/NVIDIA/gdrcopy/issues/197#issuecomment-833277121), is there any tolerance for version difference between these installations? E.g. v2.4.3 drivers in the host with v2.5 library in the container image?

Thanks in advance! 🙏

pakmarkthub, 2025-07-07:
gdrdrv (the GDRCopy driver) and libgdrapi are forward and backward compat. You can mix them. However, the features we support are the minimal set. Some unit tests in `gdacopy_sanity` may fail as they test the latest features. But you should be able to run the other applications.

drossetti, 2025-09-15:
@eicherseiji could you please close this issue?

closed 2025-09-15 by eicherseiji